madhephaestus/lx16a-servo #1: wiring and running examples

mbuyambaf, 2020-05-22:
when running the arduino sketch I run into a number of errors: first with the serial.printf but more importantly "Error compiling for board Arduino Uno.". Is there anything further I should do before I try to compile the sketch?

madhephaestus, 2020-05-22:
The Uno only has one serial port, so Serial1 would not be defied (the serial port used for the servo coms in the example). You might want to try defining a SoftwareSerial port and using that instead:

https://www.arduino.cc/en/Reference/softwareSerial

Also don’t forget you will need the line driver! https://github.com/madhephaestus/lx16a-servo#electrical

mbuyambaf, 2020-05-27:
**Okay so this is the example code that I tried:**

`#include <Arduino.h>
#include <lx16a-servo.h>
LX16ABus servoBus;
LX16AServo servo(&servoBus, 1);

void setup() {
	servoBus.begin(&Serial,
			1,// on TX pin 1
			2);// use pin 2 as the TX flag for buffer
	Serial.begin(115200);
	servoBus.debug(true);
	Serial.println("Beginning Servo Example");
}

void loop() {
	servo.move_time(0, 3000);
	delay(3000);
}`

**And this is the error that I got:** 

Arduino: 1.8.12 (Windows Store 1.8.33.0) (Windows 10), Board: "Arduino Uno"

In file included from D:\Work\xArm Controller\lx16a-servo-master\lx16a-servo-master\examples\lx16aServoExample\lx16aServoExample.ino:2:0:

C:\Users\fkm00\Documents\Arduino\libraries\lx16a-servo\src/lx16a-servo.h:172:14: warning: overflow in implicit constant conversion [-Woverflow]

  int _baud = 115200;

              ^~~~~~

C:\Users\fkm00\Documents\Arduino\libraries\lx16a-servo\src/lx16a-servo.h: In member function 'void LX16ABus::begin(HardwareSerial*, int, int)':

C:\Users\fkm00\Documents\Arduino\libraries\lx16a-servo\src/lx16a-servo.h:71:20: error: 'OUTPUT_OPENDRAIN' was not declared in this scope

   pinMode(myTXPin, OUTPUT_OPENDRAIN);

                    ^~~~~~~~~~~~~~~~

exit status 1
Error compiling for board Arduino Uno.

**Here is a an image of the wiring:**

[image]

Is there an issue with this wiring?

madhephaestus, 2020-05-27:
no i just never checked this on standard arduino. Ill need to retool the library a bit. since it uses Serial as a debug channel :/

mbuyambaf, 2020-05-27:
Okay, sure. I am more than willing to test out the code once you have made the changes.

madhephaestus, 2020-05-29:
im still looking into this, but would you mind letting me know: How did you hear about this library, i *just* finished it and haven't done any advertisement for it...

mbuyambaf, 2020-05-30:
I found it a comment you made on newscrewdriver.com attaching a link to this page.

commit 79969cfb (2020-04-21): Adding electrical information to close #1
closed 2020-09-22 by madhephaestus