madhephaestus/lx16a-servo #3: Debug board

matt4545, 2020-06-26:
Any reason why this code wouldn't work with a debug board (Using TX RX pins from debug board to TX RX pins on MCU (Teensy 4.1))? Thanks!

stevewells20, 2020-07-07:
I've been trying to use the recommended debug board with a K210 as MCU, and no luck. Not sure how that interface would work with this library, which is unfortunate because this seems really well built.

madhephaestus, 2020-09-22:
the teensy supports open collector mode for tx and rx. try just wiring the motor without the adapter board. See the readme for the wiring instructions.

madhephaestus, 2020-09-22:
Also, take a look at these wireing instructions, dont use the "debug board" as the line driver, use the $0.43 chip.  Here are detailed instructions for how to wire up the Lewansoul motors using the $0.43. https://github.com/Hephaestus-Arm/HephaestusArm2/blob/0.1.1/electronics.md#2-setting-up-the-board

closed 2021-01-18 by madhephaestus