madhephaestus/lx16a-servo #4: Debuging problem

felixNilo, 2020-09-22:
I'm testing the servo with the buslinker v2.2 as it said in the owners manual. I'm using an arduino DUE to get comm with the board, but when i debug the program I'm only getting this message from the library:
[image]

I tried to get some data from the servo board but it seems that there is no data coming.

The wiring:
[image]
[image]

White dupont is on 5v and orange is on gnd.

The code that I'm testing:

```
#include <Arduino.h>
#include <lx16a-servo.h>
LX16ABus servoBus;
LX16AServo servo(&servoBus, 2); //Servo id given by the software Bus Servo Terminal

void setup() {
	servoBus.begin(&Serial1,
			18,// on TX pin 14
			7);// use pin 7 as the TX flag for buffer
	Serial.begin(115200);
	servoBus.debug(true);
	Serial.println("Beginning Servo Example");
}

void loop() {
	int divisor =4;
	for (int i = 1; i < 1000/divisor; i++) {
		long start = millis();
		uint16_t angle = i * 24*divisor;
		int16_t pos = 0;
		pos = servo.pos_read();
   Serial.println(pos);
   Serial.println(angle);
   Serial.println(servo.isCommandOk());
		Serial.print("\n\nPosition at " + String(pos) +
				servo.isCommandOk() ? "OK" : "\n\nERR!!\n\n");

		do {
			servo.move_time(angle, 10*divisor);
		} while (!servo.isCommandOk());
		Serial.print("Move to " + String(angle) +
				servo.isCommandOk() ? "OK" : "\n\nERR!!\n\n");
		Serial.println("Voltage = " + String(servo.vin()));
		Serial.println("Temp = " + String(servo.temp()));
		Serial.println("ID  = " + String(servo.id_read()));
		Serial.println("Motor Mode  = " + String(servo.readIsMotorMode()));
		long took = millis()-start;
		long time = (10*divisor)-took;
		if(time>0)
			delay(time);
		else{
			Serial.println("Real Time broken, took: "+String(took));
		}
	}

	servo.move_time(20, 3000);
	delay(3000);
}

```

It has to be mentioned that through the software Bus Servo Terminal I'm able to make work the Servo quite well.

Heelp pleease!!

madhephaestus, 2020-09-22:
I never tested using the buslinker, and always used https://www.digikey.com/product-detail/en/texas-instruments/SN74HC126N/296-8221-5-ND instead. Here are detailed instructions for how to wire up the Lewansoul motors using the $0.43 chip linked above. https://github.com/Hephaestus-Arm/HephaestusArm2/blob/0.1.1/electronics.md#2-setting-up-the-board

felixNilo, 2020-09-22:
I will try yo buy the chip but it will take long time to have it with me, so for now I need to work the buslinker. Any idea to make it work?

madhephaestus, 2020-09-22:
TBH, i dont even own one, so i can t really test. Do you have an o-scope that you can use to look at the traces? If so put one channel on the tx enable, and the other on the tx, while running a command, capture that image and post it. do the same for the RX line. Then do the same for the serial pin on the white header (the conditioned signal to the motors) It might help me understand what its doing differently than the raw line driver chip.

felixNilo, 2020-09-22:
ASAP I will get the signals from the outputs channels of the buslinker. By now I'm keeping researching to get this work.
When I get the images i will post it.
Regards.

felixNilo, 2020-09-22:
@madhephaestus I have solved the problem using the oficial documentation.
The ino that I used is the following:
`
#define GET_LOW_BYTE(A) (uint8_t)((A))
//Macro function  get lower 8 bits of A
#define GET_HIGH_BYTE(A) (uint8_t)((A) >> 8)
//Macro function  get higher 8 bits of A
#define BYTE_TO_HW(A, B) ((((uint16_t)(A)) << 8) | (uint8_t)(B))
//put A as higher 8 bits   B as lower 8 bits   which amalgamated into 16 bits integer

#define LOBOT_SERVO_FRAME_HEADER         0x55
#define LOBOT_SERVO_MOVE_TIME_WRITE      1
#define LOBOT_SERVO_MOVE_TIME_READ       2
#define LOBOT_SERVO_MOVE_TIME_WAIT_WRITE 7
#define LOBOT_SERVO_MOVE_TIME_WAIT_READ  8
#define LOBOT_SERVO_MOVE_START           11
#define LOBOT_SERVO_MOVE_STOP            12
#define LOBOT_SERVO_ID_WRITE             13
#define LOBOT_SERVO_ID_READ              14
#define LOBOT_SERVO_ANGLE_OFFSET_ADJUST  17
#define LOBOT_SERVO_ANGLE_OFFSET_WRITE   18
#define LOBOT_SERVO_ANGLE_OFFSET_READ    19
#define LOBOT_SERVO_ANGLE_LIMIT_WRITE    20
#define LOBOT_SERVO_ANGLE_LIMIT_READ     21
#define LOBOT_SERVO_VIN_LIMIT_WRITE      22
#define LOBOT_SERVO_VIN_LIMIT_READ       23
#define LOBOT_SERVO_TEMP_MAX_LIMIT_WRITE 24
#define LOBOT_SERVO_TEMP_MAX_LIMIT_READ  25
#define LOBOT_SERVO_TEMP_READ            26
#define LOBOT_SERVO_VIN_READ             27
#define LOBOT_SERVO_POS_READ             28
#define LOBOT_SERVO_OR_MOTOR_MODE_WRITE  29
#define LOBOT_SERVO_OR_MOTOR_MODE_READ   30
#define LOBOT_SERVO_LOAD_OR_UNLOAD_WRITE 31
#define LOBOT_SERVO_LOAD_OR_UNLOAD_READ  32
#define LOBOT_SERVO_LED_CTRL_WRITE       33
#define LOBOT_SERVO_LED_CTRL_READ        34
#define LOBOT_SERVO_LED_ERROR_WRITE      35
#define LOBOT_SERVO_LED_ERROR_READ       36

#define LOBOT_DEBUG 1  /*Debug ：print debug value*/

byte LobotCheckSum(byte buf[])
{
  byte i;
  uint16_t temp = 0;
  for (i = 2; i < buf[3] + 2; i++) {
    temp += buf[i];
  }
  temp = ~temp;
  i = (byte)temp;
  return i;
}

void LobotSerialServoMove(HardwareSerial &SerialX, uint8_t id, int16_t position, uint16_t time)
{
  byte buf[10];
  if(position < 0)
    position = 0;
  if(position > 1000)
    position = 1000;
  buf[0] = buf[1] = LOBOT_SERVO_FRAME_HEADER;
  buf[2] = id;
  buf[3] = 7;
  buf[4] = LOBOT_SERVO_MOVE_TIME_WRITE;
  buf[5] = GET_LOW_BYTE(position);
  buf[6] = GET_HIGH_BYTE(position);
  buf[7] = GET_LOW_BYTE(time);
  buf[8] = GET_HIGH_BYTE(time);
  buf[9] = LobotCheckSum(buf);
[... 255 lines cut ...]
  return ret;
}

void setup() {
  // put your setup code here, to run once:
  Serial.begin(9600);
  Serial.println("Starting servo");
  Serial1.begin(115200);
  delay(1000);
}

#define ID2   2

void loop() {
  // put your main code here, to run repeatedly:
  LobotSerialServoMove(Serial1, ID2, 500, 500);
  delay(1000);
  LobotSerialServoReadPosition(Serial1, ID2);
  delay(1000);
  LobotSerialServoMove(Serial1, ID2, 700, 500);
  delay(1000);
  LobotSerialServoReadPosition(Serial1, ID2);
  delay(1000);
}
`

closed 2020-09-22 by felixNilo