madhephaestus/lx16a-servo #8: Error: no matching function for call to 'LX16AServo::LX16AServo(LX16ABus&, int)'

daxliniere, 2022-01-03:
Hey Kevin,
Happy new year.

I'm trying to get your library to work with LX-16a and LX-224 with ESP32 (using pin 4 for data). I've triple-checked the pinouts and everything is correct, including 3.3v measured on data pin.

When I try the One-Wire sketch (pin 33 changed to 4), I repeated get "Read TIMEOUT Expected 7 got 0 on cmd: 21 id 1 took 30" in serial monitor.

When I try the code below (pin 33 changed to 4), I get the error in Arduino IDE: "no matching function for call to 'LX16AServo::LX16AServo(LX16ABus&, int)' "
```
#include <Arduino.h>
#include <lx16a-servo.h>
LX16ABus servoBus;
LX16AServo servo(servoBus, 1);
int n = 0;
void setup(){
  servoBus.begin(Serial1, 4);
}

void loop(){
    uint16_t angle = (n%11) * 100;

    uint8_t params[] = { (uint8_t)angle, (uint8_t)(angle>>8), 500&0xff, 500>>8 };
    bool ok = servo.write(1, params, sizeof(params));
    printf("Move to %d -> %s\n", angle, ok?"OK":"ERR");

    delay(10);

    ok = servo.read(2, params, 4);
    printf("Position at %d -> %s\n", params[0]|(params[1]<<8), ok?"OK":"ERR");

    n++;
    delay(2000);
}
```

Could you point me in the right direction for how to get this working, please?

Thank you very much and all the best for 2022,
-Dax.

madhephaestus, 2022-01-05:
ESP32 has its own initializer: https://github.com/madhephaestus/lx16a-servo/blob/master/examples/lx16aOneWireESPExample/lx16aOneWireESPExample.ino#L9

You also need to pas a pointer to the serial object, not the reference syntax. You dropped out the '&' ahead of the Serial1.

closed 2022-01-05 by madhephaestus

daxliniere, 2022-01-05:
Thanks, Kevin! Also, does your code talk to the servos directly, or do you still need to use LewanSoul's TTL board?

madhephaestus, 2022-01-05:
it talks directly! the ESP supports single pin serial so the TX/RX pin state switching most other microcontrollers need is managed in the IO crossbar inside the chip, and controlled by the library when in onePinMode

madhephaestus, 2022-01-05:
> Thanks again, Kevin. Yeah, that's what I thought, but just not having any
> luck with it for some strange reason.
> I've just tried again using your updated code, infact, I tried all of your
> example sketches, but cannot get the LX-224 or LX-16A to move. Power, gnd
> and signal are all 100% correctly connected. Power is 6.6v, signal is
> showing 3.27v, both measured at the servo itself.
> I have also tried using *LX16AServo servo(&servoBus, LX16A_BROADCAST_ID)*
> to no avail.
>
> I ran the Set ID sketch and set it to ID3, then back to ID1, but can't get
> anything from the one-wire example except "*Read TIMEOUT Expected 7 got 0
> on cmd: 21 id 254 took 30*" in Serial Monitor.
>
> For the Set ID sketch, I changed setup to:
>
> void setup() {
>   servoBus.beginOnePinMode(&Serial1,4);
>   servoBus.debug(true);
>   servoBus.retry=0;
> 	Serial.begin(115200);
> }
>
> At this point, I"m completely stuck. :(

Maybe check that you have the connector the right way round?

Try switching pins to 33?

6.6v sounds low, they like 7.5-9v the hv ones take 12v.

Do you have an o-scope to look at the signal?

daxliniere, 2022-01-05:
Hey Kevin! I just bumped it up to 8v and now the one-wire sketch gives me something different! Alriiight!

Still no movement, but this seems like a great step forward.

Now getting:
**ERROR Capped set at min 105144 attempted 5000
Setting ticks 1000 on ID 1**

I'll try the motor mode sketch next.

daxliniere, 2022-01-05:
It spins! Motor mode works!
Thank you so much for your help, Kevin. I should be able to work it out from here. I just had no way of knowing if the servo was even communicating with the ESP32.

All the best!
Dax.

daxliniere, 2022-01-05:
If you have any tips on what those errors above mean, I'd be very grateful. :)

madhephaestus, 2022-01-18:
it looks like somewhere along the way the min and max limiting in teh motor got set to some junk values. Try using calibrate() to set a known value and range explicatly. FYI the encoder in the motors is just a POT, so it only has sensing range of 240 degrees. whe the motor is in any other range (the 120 deg of unencoded range) the read position may come back with nothing but junk values. i think that junk might have been read and sent back down as limits...