madhephaestus/lx16a-servo #10: Servo not moving

SameemN, 2022-10-08:
I am using an esp32 and have wired it as you said but when I upload the one-pin example, nothing happens.

madhephaestus, 2022-10-08:
how are you powering it? it draws more current then a USB port can source. 
Are you sure the serial cable is not reversed? its super easy to get that backwards. 
Have you set the motor ID? what ID is the motor connected, and what ID are you talking to?

SameemN, 2022-10-08:
I'm powering it with a separate power supply, the Id is 3 and I believe I have set it to 3 in the example. I am unsure what the 33 means in the begin in one pin mode   line, does that have something to do with it?

madhephaestus, 2022-10-08:
thats the pin number for the serial line

madhephaestus, 2022-10-08:
the serial print outs are a very important part of the examples. what is being printed when setting the ID? what is printed when running?

SameemN, 2022-10-08:
That's what I figured it was but was unsure. The servo is still not moving.

madhephaestus, 2022-10-08:
going to need all the print outs i asked for above, and a detailed photo that shows the wireing clearly in frame. I should be able to trace all wires and how they are connected.

SameemN, 2022-10-08:
ok here you go 
[image]
[image]
[image]

SameemN, 2022-10-08:
I set the id using the debugging board for the servos

madhephaestus, 2022-10-08:
its not communicating in that printout, that means an electrical or an addressing problem. Did you see it print on the serial that it set the ID ok?

check you have the servo pins correct:

https://github.com/Hephaestus-Arm/HephaestusArm2/blob/0.1.1/photos/motor_cable.jpg

check the voltage, it can not communicate without servo power. 

Also check you have the esp32 ground connected to the power supply ground.

madhephaestus, 2022-10-08:
(Side note) That's an awesome looking robot design! I would love to collect images of projects that use this library, please drop a message (a PR to add it to the readme would work) so i can help highlight cool projects like yours!

SameemN, 2022-10-08:
I double-checked everything, the servo is receiving power. The servo connections appear to be correct. I ran the example to set the servo ID and it says this: 
[image]

SameemN, 2022-10-08:
Also thanks! I'm not super familiar with GitHub so am unsure what you mean by PR. Thanks for all the help btw!

madhephaestus, 2022-10-08:
I made an update to the set id code so it will show errors if the ID set fails. Run the new id set example and show me it setting the ID. Dont forget to change the ID set initialization to use your hardware configuration.

SameemN, 2022-10-08:
It is saying the ID set failed.

HafthorArni, 2022-10-14:
I also noticed with an oscilloscope while running the original one wire ESP example code that pin 33 is pulled up to 3.3V but no signal is being sent through there, the signal is sent through pin TX0.
If I also connect this wire to the servo there is still no movement but the serial monitor changes from displaying
 "Read TIMEOUT Expected 7 got 0 on cmd: 21 id 1 took 30" 
to This:

[image]

madhephaestus, 2022-10-26:
Note that the set ID code and the one wire mode code have 2 different setup code. It looks like you left the id set code unchanged, and so tried to use one of the programming pins as your serial bus pins (which will cause programming to die in the middle)

rgunch2, 2022-10-28:
I am having the same experience as SameemN. Everything appears to be wired correctly and I have changed the set ID setup code to match my hardware, but it is failing to set the ID. I'm also getting the same error in the OneWire code, I assume because the ID is not set properly. When running the set ID code with the setup changed to servoBus.beginOnePinMode(&Serial2,33), pin 33 remains high at 3.3V with no activity on the oscilloscope. I've tried several boards as well to see if it was a hardware issue, with no success.

HafthorArni, 2022-10-28:
This is the exact same thing I did also^ 
I did change the set ID code as you can see in the last screenshot in my first comment. Maybe it's unclear because I only commented out the original hardware initialization and then added the servoBus.beginOnePinMode(&Serial2,33).
In this case the servo signal pin was connected to pin 33. I tried a few different servos and two different ESP32.
Does the servoBus.debug(True) and servoBus.retry = 0 also need to be in there?

madhephaestus, 2022-10-28:
Try a different pin? Maybe your module is using pin 33? I use the ESP32 Devkit-C v4 and pin 33 works. Here is my firmware that uses 3 servos: https://github.com/Halloween2020TheChild/FirmwareTheChild/blame/main/FirmwareTheChild.ino

leohafi, 2023-03-21:
@SameemN @rgunch2 @HafthorArni 
have you found a solution?
I have the same problem if try running the sample files i can´t get them to work with one wire modes ESP32 (i have esp32doit-devkit-v1). I changed the Setup code and already tried different pins and boards. Also checked with oscilloscope and it´s flat 3.3V.
Or is there maybe smth not working with open-drain mode setup?
thx for help :)

madhephaestus, 2023-10-01:
i think the issue you are having is realated to #15 and its resolution in the README. you may just need to downgrade the toolchain to an older version.