madhephaestus/lx16a-servo #15: Servo drivers will not work unless you use the correct ESP32 toolchain

migsdigs, 2023-09-24:
Previously I was working with a Hiwonder xArm with 6 servos (5 lx15-d servos and a lx-225) and I had managed to get these up and running with a NodeMCU-32s v1.3 esp32 dev board. My dev environment is PlatformIO and I was making use of the toolchain version 2.0.0 (set in the platformio.ini file as platform = espressif32@2.0.0).

This dev board (NodeMCU-32s) is no longer available, so I have acquired an Espressif ESP32-DevKitC-32E board. I configured my PlatformIO environment for this board, i.e. I made use of a updated toolchain (4.0.0 I believe). With this environment, I was unable to read or write to the servos, with them throwing errors whenever I tried to access them.

I found that I was able to get around this by using my original dev and board configuration that I was using for the NodeMCU board. It should be notes that the pin layout of the two boards is also more or less the same. I used the following configuration in my PlatformIO ini file:

```
[env:nodemcu-32s]
platform = espressif32@2.0.0
board = nodemcu-32s
framework = arduino
monitor_speed = 115200
lib_deps = 
    madhephaestus/lx16a-servo@^0.9.3
```

madhephaestus, 2023-09-24:
thank you for adding this!

I think this is a good enough tip that we should add it to the README.md file. Can you make a PR with this content added to the readme so folks coming to the main page will see it?

migsdigs, 2023-10-01:
> thank you for adding this!
> 
> I think this is a good enough tip that we should add it to the README.md file. Can you make a PR with this content added to the readme so folks coming to the main page will see it?

Done. Feel free to make changes

commit 7e428543 (2023-10-01): Merge pull request #16 from migsdigs/patch-1

Update README.md to include a recommended environment setup

Thank you!

close #15
closed 2023-10-01 by madhephaestus