madhephaestus/lx16a-servo #18: Arduino Compatibility

DomGravante, 2024-08-17:
Hi, thanks for providing this library. I'm having some difficulties getting any unit to spin a LX-224 servo. I've tried all the examples and lots of the troubleshooting in other threads / Issues. 

I've noticed that this library appears to be written for a certain ESP-32 board, and is not intended to be generally compatible with all Arduinos. Can you share which board(s) you've tested and had this run successfully? I've had no success on a Uno R3 (makes sense - not an ESP-32 board) and no success on my Nanos which, to my understanding are ESP-32 based. 

For example, I notice `Serial1` pointers in the example code, which is only applicable for a subset of arduino boards. Is this on purpose? Do you have any recommendations on how you would approach troubleshooting this?

DomGravante, 2024-08-17:
After some further reading, I realized there's two flavors of Nano - an ESP32 variant and a ATmega328P variant. It looks like I have the ATmega328P. To your knowledge, is there anything that would prevent this library from being used on boards such as the Uno R3 & Nano ATmega328P, instead of a ESP32 based board? I can't imagine the Serial communication would be different, but maybe the compilation?

madhephaestus, 2024-08-17:
you need to use a buffer chip on any arduino besides the ESP32 (using the 1.0.4 toolchain only). This motor type uses a tx and rx on a single pin, changing from high impedence reciver, to low impedence transmitter. the only way to make a normal arduino use a single pin in both modes is to use a buffer chip to manage the switch from low to high impedence.

closed 2024-08-17 by madhephaestus